riscv-non-isa/riscv-server-soc #12: Standardize the UART in SoC

ShawnLess, 2023-10-12:
Hi all,

UART plays a basic but critical role in system development and early bring up, e.g., BSP input/output,  early console in linux, shall we standardize the UART for RISC-V SoC, referring the de faco standards such as 16550 compatible UART. (ARM uses its own PL011 anywhere).

Standardizing UART across different platform should reduce overall BSP/Kernel porting effort.

ved-rivos, 2023-10-13:
The BRS specification requires a [SPCR](https://github.com/riscv-non-isa/riscv-brs/blob/main/acpi.adoc#spcr) for early OS boot console if a UEFI protocol supported graphical console is not present. We could add a requirement to support a 16550 compliant serial port when graphical console is not present.

ShawnLess, 2023-10-17:
That would be great!

ved-rivos, 2023-10-18:
Added UART requirement in PR #14

closed 2023-10-24 by ved-rivos