riscv-non-isa/riscv-server-soc #16: Preserve port configuration registers in DL_Down status.

ShawnLess, 2023-11-07:
In PCIe Spec 6.0, Chapter 2.9.1, 

"For a Downstream Port, DL_Down status is handled by:
• Initializing back to their default state any buffers or internal states associated with outstanding requests transmitted Downstream
      **Port configuration registers must not be affected, except as required to update status associated with the transition to DL_Down**."

Most of the implementation will "Initializing back to their default state" during DL_Down status, but ignores the "Port configuration registers must not be affected".

This leads to serious problem in hot-plug scenario, where the link related registers (max payload size, link width/size information) are lost after removing the device, and won't be able to recover after the devices is plugged back.

Wondering shall we specifically adding this rule.

ved-rivos, 2023-11-13:
Losing the port configurations is clearly violating the PCIe specification. However if you have observed this is violated some implementations then it makes sense to add a rule about this.

ved-rivos, 2023-11-13:
PR #17 adds requirement AER_080 to retain port  configuration on transition to DL_Down.

closed 2023-11-15 by ved-rivos