riscv-non-isa/riscv-server-soc #19: IMSIC/MRIF memory regions

dkruckemyer-ventana, 2023-11-22:
It's kinda obvious from the definition of the IMSIC interrupt files in the AIA, but should requirement IIC_070 (which defines the IMSIC interrupt file PMAs) also include non-idempotent? What about the ordering PMA? Stating both non-idempotent and I/O channel 0 strong ordering (for ordering) would bring the definition inline with the PBMT IO mode, for whatever that's worth.

Also, does there need to be a requirement for MRIF PMAs? I'm not sure whether software would want MRIFs to be cacheable or not, but if there's a preference (or even if the behavior is system specific), we should call it out. Certainly, the minimum level of AMO and/or Reservability PMAs should be specified, since the IOMMU must support atomic updates of the MRIF.

ved-rivos, 2023-12-01:
> Stating both non-idempotent and I/O channel 0 strong ordering (for ordering) would bring the definition inline with the PBMT IO mode, for whatever that's worth.

Agree.

> Certainly, the minimum level of AMO and/or Reservability PMAs should be specified, since the IOMMU must support atomic updates of the MRIF.

The AIA specification requires AMOOR operation and thus AMOLogical level support. Software should be programming the MRIF from memory that supports AMOLogical. The IOM_250 requires the host bridge to check physical memory attributes for memory accesses made by the IOMMU.

ved-rivos, 2023-12-06:
Updated in https://github.com/riscv-non-isa/server-soc/pull/26/commits/331ee4a5c38fe26229f893701a320a1dae2e9c89

closed 2023-12-13 by ved-rivos