riscv-non-isa/riscv-server-soc #20: Many requirements are out of scope with respect to the charter

darius-bluespec, 2023-11-22:
The current charter states that

> The specification standardizes the requirements for the hardware interfaces and capabilities [...] provided by the SoC to software executing at privilege level less than M

... which has generally been interpreted to mean that only behavior that is visible in S or HS/VS modes is in scope.

There are many requirements, in particular most or all of the Manageability, Debug, and Trace sections, but also miscellaneous requirements in other sections, are not visibile in the way defined above.

This is highlighted in the test plan, in which the entire Manageablity and Trace sections are deemed untestable, and the Debug section requires tests to be executed on "a debug host connected to Server SoC".  While there are potentially many reasons a particular requirement may be untestable (obscure error conditions, requirement for impractical test harness, etc.), those requirements that are untestable *specifically* because they are not visible from a privilege level less than M are in fact at odds with the current charter as written.

ved-rivos, 2023-11-22:
For the manageability - the test spec is WIP - the I2C/I3C interface and redfish interface for providing communication with the management controller. These are visible as for in-band management of the platform. You highlight a good point that the quoted statement should be expanded to include the hardware capabilities that would be required for developing/debugging standard operating systems on a platform. Standard operating systems work on a platform because someone takes the time and effort to make that happen and the platform should provide the right tools for someone to make it happen.

darius-bluespec, 2023-11-22:
True for some of Manageability, but MNG_030 and MNG_040 are by definition not visible from a privilege level less than M.

ved-rivos, 2023-12-06:
MNG_030 - the UART will be used by the OS as the console.
MNG_040 - agree its not SW visible.

ved-rivos, 2023-12-13:
Updates in https://github.com/riscv-non-isa/server-soc/pull/27

closed 2023-12-15 by ved-rivos