riscv-non-isa/riscv-server-soc #22: SID_080: RCiEP MAY support the PCIe enhanced allocation (EA) capability for fixed allocation of memory resources.

andreiw, 2023-12-01:
Is there a benefit to allowing this? EA is not really an industry-accepted practice in the server space. I am aware of it only being used by Cavium ThunderX CN88xx processors and subsequent Cavium/Marvell Octeon Si. Handling EA is a corner case with respect to PCIe resource handling, pass-through, etc and not all OSes support EA.

ved-rivos, 2023-12-01:
There is benefit in allowing this capability. This  is a standard PCIe capability supported in PCIe Gen 6. The EA capability supports providing memory resources for PF and VFs. For integrated devices which are not plug-n-play and don't need their resources resized/moved this provides a simple and convenient integration option.

andreiw, 2023-12-06:
How is OS support for EA? It's hardly typical. I've implemented EA in ESXi but this was limited to the subset exercised by Cavium silicon - ie. no resources without a BEI, no W (non HWInit) resources and my understanding is that it's seen no adoption in Windows, so if I were a Si vendor aiming for most compatibility, I'd avoid EA unless I had a really good reason.

Can we restrict the subset of EA? E.g. avoid 'W'(ritable) entries, avoid entries without a BEI, avoid bridge resources (IIRC the Cavium  implementation had switches without any resources, as if they were subtractive). It's unclear that anyone implementing EA today won't run into massive OS support investment.

andreiw, 2023-12-07:
That sounds reasonable!

ved-rivos, 2023-12-07:
Update in https://github.com/riscv-non-isa/server-soc/pull/26/commits/ef65d1cdc4159ac7d3e2035620e90c598cc4bb5c

closed 2023-12-10 by ved-rivos