riscv-non-isa/riscv-server-soc #24: PCIe integration: do we need rules around aligned writes?

andreiw, 2023-12-01:
See https://developer.arm.com/documentation/den0029/latest

RS_PCIe_03 The Root Complex must:
• Send 2B PE writes that are 2B aligned as 2B PCIe writes.
• Send 4B PE writes that are 4B aligned as 4B PCIe writes.
• Send 8B PE writes that are 8B aligned as 8B PCIe writes.

RS_PCIe_04 The System must ensure that:
• Aligned 2B writes from Endpoints reach the target as 2B writes.
• Aligned 4B writes from Endpoints reach the target as 4B writes.
• Aligned 8B writes from Endpoints reach the target as 8B writes.

ved-rivos, 2023-12-01:
The ECM_010 and MMS_030 were intending to do that. But they can be made more explicit. For ECM_010 will extend the rule to state "One, two, and four byte naturally aligned read and write MUST be supported and lead to a single CfgRd*/CfgWr* transactions of the corresponding size.". For MMS_030, will add "Naturally aligned read and writes of size up to 8 bytes MUST result in a single MemRd/MemWr transactions of the corresponding size."

ved-rivos, 2023-12-06:
Updated in https://github.com/riscv-non-isa/server-soc/pull/26/commits/ee1361d9b4680a6729b65b25380635ca48bc1ff0 and https://github.com/riscv-non-isa/server-soc/pull/26/commits/220a1b9800c8067be2b7a5182ef03f881e66311e

andreiw, 2023-12-07:
Lgtm

closed 2023-12-07 by andreiw