riscv-non-isa/riscv-server-soc #43: Clarify requirement for IOMMU

jones-drew, 2024-06-04:
The non-normative text for IOM_010 states "The number of IOMMUs implemented in the SoC is UNSPECIFIED." which implies zero is OK. However, IOM_020 states "All DMA capable peripherals (RCiEP and non-PCIe devices) and all PCIe root ports accessible by software on the RISC-V application processor harts MUST be governed by an IOMMU." Assuming a server SoC will always have at least one DMA capable peripheral means the spec effectively mandates an IOMMU. There has been some confusion, though, from only reading IOM_010. Should we instead write something like "The number of IOMMUs implemented in the SoC is UNSPECIFIED but MUST be at least one."?

jones-drew, 2024-06-04:
Since I'm suggesting changing non-normative text, then I probably shouldn't be using such formal-looking language and 'MUST'. Maybe something more like "The number of IOMMUs implemented in the SoC is UNSPECIFIED, but is expected to be at least one." is better?

ved-rivos, 2024-06-04:
How about:
"The number of IOMMUs implemented **to satisfy requirement IOM_020** in the SoC is UNSPECIFIED."

jones-drew, 2024-06-04:
> How about: "The number of IOMMUs implemented **to satisfy requirement IOM_020** in the SoC is UNSPECIFIED."

Works for me. Thanks

ved-rivos, 2024-06-04:
update in #44

closed 2024-06-04 by ved-rivos