riscv-non-isa/riscv-server-soc #45: Server SoC spec applicable to a VM environment?

andreiw, 2024-06-13:
sunilvl raised a similar (https://github.com/riscv-non-isa/riscv-server-platform/issues/7) question for server platform. It is expected and a good thing for VM envo to mimic the hardware environment as closely as possible, but some of the requirements (VS interrupt files, IOMMU, ATS, etc) make it unrealistic that VM environments will be compliant. Perhaps such requirements could be predicated on the availability of HS mode to the system software?

Having a dedicated VM spec seems unnecessary (and with sufficiently advanced hardware and hypervisors, a VM environment would be indistinguishable in terms of features anyway).

Thoughts?

ved-rivos, 2024-06-13:
The server SoC specification provides the specification for the SoC - the physical machine.  The server SoC specification does not place any requirements on the OS that executes and how it creates virtual machine. There isnt any aspect of the server SoC specification that would preclude a VMM from emulating a virtual server SoC for a guest VM if it chooses to - including providing an emulation for the IOMMU, emulated ATS, multiple VS level interrupt files. 

From that perspective I agree that a dedicated VM specification is not required.

andreiw, 2024-06-14:
The current spec doesn't preclude, but it does mean that a VM environment would only meet certification *if* it implemented the whole spec, including aspects that are a) heavy weight b) not necessarily interesting for most of VM environments, etc,

ved-rivos, 2024-06-14:
The certification for this specification is a certification of the Server SoC hardware. It is not a certification of a software execution environment such as a virtual machine.

andreiw, 2024-06-17:
Ok so you're saying we should have a separate spec for virtual VM soc, even if such spec would be a perfect subset of the server soc spec?

ved-rivos, 2024-06-17:
I am not saying we need a separate spec. We can have a section in the server SoC spec with a table that maps requirements mandatory for a VM.

andreiw, 2024-06-17:
Ok. Would we be able to define those VM requirements as being a subset of the the server platform requirements, including as a subset of server soc requirements? If I understood correctly, relaxing a spec is not allowed.

ved-rivos, 2024-06-20:
This question was discussed [1] in the Server Platform TG. Here is a summary:
- Ved proposed that VM support could be looked in terms of separate certification, not as VM-specific lists of requirements (e.g. in SoC or platform docs)
- Anup suggested VM support could be an entirely different spec.
- Redbeard asserted VM certification to be of paramount importance.
- A few folks questioned the benefit of VM envo standardization/certification entirely. As a counter-argument Andrei mentioned SystemReady VE certification by Arm hypervisor VMs, sharing the benefit of a single binary distribution with the hardware.
- Heinrich mentioned that it is unlikely we have enough time to worry about VMs and that we ought to focus on the server platform for the time being. There were no dissenting opinions to this.
- ...as it stands, given the variance in approaches/opinions and limited time, the current take-away is to table the discussion to a more opportune time.

To ensure that Further discussion on this topic continue in one place - in [2] and on [3] - closing this issue.

[1] https://lists.riscv.org/g/tech-server-platform/message/98
[2] https://github.com/riscv-non-isa/riscv-server-platform/issues/7
[3] https://lists.riscv.org/g/tech-server-platform

closed 2024-06-20 by ved-rivos