riscv-non-isa/riscv-server-soc #46: MSI_030 disallows virtual wire (no legacy PCI devices period?)

andreiw, 2024-06-13:
I was just curious why MSI_030 disallows virtual wire, instead of leaving this as an implementation option. What if someone wants to build a system that will support legacy PCI devices (eg. hanging these from a PCIe-to-PCI switch)? Surely such a system could be compliant.

When disallowing, it's unclear what the expected behavior is when the RP receives a virtual wire TLP. Is that a PCIe error? Or nothing happens?

andreiw, 2024-06-14:
Nevertheless, the spec disallows this outright, instead of simply leaving it to the implementer to worry about implementation details and gaps. That's the part that seems off. Lack of need usually means lack of specification, not an outright ban. And if there is an outright prohibition, I think some more text is necessary to describe what happens on the unsupported virtual wire (or is that already covered by the PCI-SIG specs?)

ved-rivos, 2024-06-14:
The treatment of unsupported requests is covered by the PCI-SIG specifications.

closed 2024-06-14 by andreiw