riscv-non-isa/riscv-server-soc #58: Public Review Feedback for RISC-V Server SoC Specification

anpar123, 2024-12-10:
(Posting on behalf of different contributors at Google.)

Please find aggregate feedback on the proposed RISC-V Server SoC Specification below:

[2.3] - Nested IO translations (i.e. two stage translations gIOVA → GPA → SPA) is becoming a must have requirement, so it may be better to call out this feature as a must. It is also preferable if the Guest(unprivileged user) can issue invalidation in the gIOVA address when we are using two-level page tables.
[2.4.4] - [ADR_040] Peer-to-peer access across devices through CPUs root-complex/IOMMU is quite common, so making this a required feature may be beneficial.
[2.4.9] -It is beneficial to consider specifying the Firmware first flows to export these errors via. an out-of-band mechanism to BMC. Such mechanisms SHOULD not have any impact on the performance and be resilient to marginal links creating large numbers of correctable errors.
[2.6] - For this comment - “The count of RCID and MCID that can be used in the SoC should scale with the number of RISC-V harts in the SoC” 

- Can we make this a requirement instead of a suggestion?
- Can we include a notion of a collection of RCID/MCID which could be monitored/controlled as a unit at a higher granularity (think monitoring/controlling a VM instead of each hart within a VM).

Can we add a requirement in section [2.8] to scale the monitoring ability to support all available RCID/MCID.
[2.6] - For this comment - “The method employed by the SoC for bandwidth throttling and control is specific to its implementation. It is advisable for the implementation to utilize a scheme that results in a deviation of no more than +/- 10 % from the target set by system software through the CBQRI interface.” 

- Can we add requirements to provide mechanisms to monitor bandwidth throttling and control irrespective of the implementation?

[2.6] - [QOS_070] and [QOS_080] - Can we add other end points like  PCIE, CXL, cross socket, die, etc.?
[2.7] - [MNG_020] - Can we add/recommend I3C (no IPMI) support as well? 
[2.8] - Please add performance monitoring for IOMMU as well. Could start with the following

- Hit/Miss rates for different page sizes 
- Hit/Miss rates separately for first and second level 
- #Memory access by IOMMU for page walk

[2.8] - [SPM_010] - Please specify cache lookup and misses for read and write traffic separately.
[2.8] - [SPM_030] - Can we add P2P traffic accounting as well?
[2.8] - [SPM_050] - Can we add separation per target, in other words specific remote memory v/s any remote memory?
[2.9] - [SEC_020] - Can we add a recommendation to support CMA/SPDM?
[2.1] - Can we define a notion of maximum permissible clock skew across the distribution network?

ved-rivos, 2024-12-15:
> [2.3] - Nested IO translations (i.e. two stage translations gIOVA → GPA → SPA) is becoming a must have requirement, so it may be better to call out this feature as a must. It is also preferable if [...]

The RISC-V IOMMU supports two-stage address translation. Two stage address translation in the RISC-V IOMMU is not optional. The RISC-V IOMMU supports two commands for page-table cache invalidation: IOTINVAL.VMA and IOTINVAL.GVMA. The IOTINVAL.VMA may be used to invalidate gIOVA. The IOTINVAL.GVMA may be used to invalidate translations cached from the second-stage page tables.

ved-rivos, 2024-12-15:
> [2.4.4] - [ADR_040] Peer-to-peer access across devices through CPUs root-complex/IOMMU is quite common, so making this a required feature may be beneficial.

Acknowledge that P2P routing through root complex is beneficial.  Supporting P2P routing through the root complex was discussed in the TG and the conclusion was that for this version of the specification it is maintained as a recommendation to provide support for implementations targeting a broad market segment. We will revisit mandating this for the next revision of the specification.

ved-rivos, 2024-12-15:
> [2.4.9] -It is beneficial to consider specifying the Firmware first flows to export these errors via. an out-of-band mechanism to BMC. Such mechanisms SHOULD not have any impact on the performance and be resilient to marginal links creating large numbers of correctable errors.

The RISC-V RAS Error-record Register interface specification supports integrating the RAS handlers with error reporting by architectures such as PCIe AER (section 2.6 of RISC-V RERI specification; standard error code 27: "PCIe/CXL detected (e.g., logged into PCIe AER, CXL.mem error log, etc.) errors"). Rules for supporting RERI is included in this specification. The rule RAS_080 requires supporting corrected error counters with counter overflow support. When error record is placed in counting mode for corrected errors the RAS handler is signaled on a counter overflow allowing the RAS handler to configure thresholds on the rate at which corrected errors are signaled.

ved-rivos, 2024-12-15:
> Can we add a requirement in section [2.8] to scale the monitoring ability to support all available RCID/MCID.

The RISC-V CBQRI specification supports using all available MCID for monitoring purposes and all available RCID for configuring capacity/bandwidth allocations.

ved-rivos, 2024-12-15:
> [2.6] - For this comment - “The method employed by the SoC for bandwidth throttling and control is specific to its implementation. It is advisable for the implementation to utilize a scheme that [...]

The RISC-V CBQRI provides a standard register interface to configure bandwidth allocation and monitor bandwidth usage irrespective of the implementation. Specifying an implementation would be out of scope for RISC-V specifications. 
 This note notes that the mechanism used for bandwidth throttling and control is implementation specific i.e., the type of signals, meters, control loops, etc. employed by the implementation to achieve the targets configured using the architectural CBQRI interface. The note is further guiding the implementation towards the error bars they should strive to achieve.

ved-rivos, 2024-12-15:
> [2.6] - [QOS_070] and [QOS_080] - Can we add other end points like PCIE, CXL, cross socket, die, etc.?

QOS_070 addresses the CXL.mem controllers. QOS_040 is addressing memory requests originated by devices - including PCIe devices. This version of the specification attempts to address the most important capacity and bandwidth controllers (significant caches and memory controllers). We will discuss expanding the rules to cover other controllers like D2D controllers and inter-socket interconnects in the next revision of the specification.

ved-rivos, 2024-12-15:
> [2.7] - [MNG_020] - Can we add/recommend I3C (no IPMI) support as well?

Yes. Will include a recommendation to support I3C.

ved-rivos, 2024-12-15:
> [2.8] - Please add performance monitoring for IOMMU as well. Could start with the following
> Hit/Miss rates for different page sizes
> Hit/Miss rates separately for first and second level
> #Memory access by IOMMU for page walk

The IOMMU performance monitoring is addressed by IOM_180 through IOM_210. The RISC-V IOMMU specification includes the following standard events  for an IOMMU: translated/untranslated/ATS requests, TLB misses, device/process directory walks, first and second stage  page table walks. Will add a cross reference from section 2.8 to the IOMMU section.

ved-rivos, 2024-12-15:
> [2.8] - [SPM_010] - Please specify cache lookup and misses for read and write traffic separately.

Agree.

ved-rivos, 2024-12-15:
> [2.8] - [SPM_030] - Can we add P2P traffic accounting as well?

Agree.

ved-rivos, 2024-12-15:
> [2.8] - [SPM_050] - Can we add separation per target, in other words specific remote memory v/s any remote memory?

We will work on additional mandates on performance monitoring for NUMA configurations in next revision of the specification.

ved-rivos, 2024-12-15:
> [2.9] - [SEC_020] - Can we add a recommendation to support CMA/SPDM?

This requirement is for the root port. While PCIe specification requires SPDM secure messaging and DOE for upstream ports and endpoints, the specification permits root complexes to support IDE KM using DOE in the root port, DOE in the RCRB, or an implementation specific mechanism. Specifically, root ports are not required to support SPDM secure messaging. For this version of the specification the TG chose not add additional requirements to root ports that are not mandated by the PCIe specification.

ved-rivos, 2024-12-15:
> [2.1] - Can we define a notion of maximum permissible clock skew across the distribution network?

This is addressed by the RISC-V unprivileged architecture. The Zihpm extension requires the real-time clocks of all harts must be synchronized to within one tick of the real-time clock.

ved-rivos, 2024-12-15:
PR #60 created to address the feedback.

ved-rivos, 2024-12-15:
@anpar123 - thank you much for the feedback. Please see comments and PR to address the feedback.

closed 2025-01-05 by ved-rivos