riscv-non-isa/riscv-server-soc #71: PCI interrupt transfer rules

chenwhJ, 2025-03-28:
Hi,
 In RISCV Server Soc spec 2.2, there are two rules for PCI interrupt transfer ordering:

1. "Specifically, interrupts issued by a device following a write
operation must be processed only after the previous write operations have been completed and observed."
2. "Similarly, interrupts issued by a device must be observed before any subsequent read completions generated by
the device."

I wonder if the first rule is for PCI MSI-style interrupt transfer and the second rule is for PCI INTx-style interrupt transfer?

Consider the scenario that PCI devices deliver writes to memory through DMA. After PCI delivered the last write requests, PCI device deliver an interrupt  to inform CPU that PCI device has completed the write transfer and CPU could take the data from memory.
    For PCI MSI transfer,  the MSI delivered by PCI device must be observed after the former PCI writes completed. This is ganranteed by PCI ordering rules (Row A, Col 2) in the table listed below
    For INTx transfer, CPU should read PCI device in interrupt handler. When PCI device returns read complete, it is ganranteed that former PCI writes has be observed in memory. This is ganranteed byPCI ordering rules (Row D, Col 2) in the table listed below.
 
[image]

 In conclusion, I thought the first rule in RISCV Server Soc spec IIC-020 refers to MSI delivery, and the second rule refers to INTx delivery. Please correct me if my thoughts are wrong, Thanks!

ved-rivos, 2025-03-28:
The scenarios you outline look right and follow the ordering rules of PCIe. 

In PCIe, the INTx virtual wire interrupt signaling mechanism is used to support legacy endpoints and PCI/PCI(-X) bridges where MSI/MSI-X cannot be used. In this case the INTx mechanism uses 8 message - TLP type Msg - to assert and deassert the four virtual wires A, B, C, and D. The Msg TLP, similar to Mwr TLP (used by MSI), is a posted TLP and will fall under column 2. So as far as ordering rules are concerned, the D2a rule applies to determining whether a Cpl TLP can pass a Mwr (even when used to signal MSI) or a Assert_INTx/Deassert_INTx Msg (used for INTx signaling). 

The Server SoC specification provides the rule IIC-020 that states MSIs must be supported. The text you quoted is informative text and are not rules of the Server SoC specification. That text is reminding that certain protocols like PCIe may use Posted TLPs (Mwr) to signal MSI and that PCIe provides the ordering rules (Table 2-42).

chenwhJ, 2025-04-01:
Thank you for your replay.

To support D2a rule, the endpoint should delivery a response to  inform that  it has received the Mwr, so that the bus interconnect could guarantee the sequence of Mwr and read completion.

 I'm considering a SoC system which integrated PCI device，but the MSI transfer path DO NOT satisfy the D2a rule ( which means that the sequence of read completion and MSI write is not guaranteed ), will this cause problem for MSI delivery?

In my opinion, there is no scenario for MSI-delivery to use the D2a rule. Thus the D2a rule may NOT be supported for MSI transfer path. The mechanism of MSI can be satisfied merely by the A2a rule.

> The scenarios you outline look right and follow the ordering rules of PCIe.
> ……
> The Server SoC specification provides the rule IIC-020 that states MSIs must be supported. The text you quoted is informative text and are not rules of the Server SoC specification. That text is reminding that certain protocols like PCIe may use Posted TLPs (Mwr) to signal MSI and that PCIe provides the ordering rules (Table 2-42).

closed 2025-04-16 by ved-rivos