riscv-non-isa/riscv-server-soc #73: Clarify requirement for IOM_130

xxy-peng, 2025-05-07:
> IOM_130 IOMMU MUST support MSI and MAY support wire-signaled interrupts for external
interrupts originated by the IOMMU itself.

[image]

MY question is:
    Does this requirement mean that external interrupts generated by the IOMMU itself MUST support use MSI (while WSI is optional), or does it imply that only PCIe device interrupts are mandated to use MSI (while IOMMU-originated interrupts allowed to use WSI and not support MSI) ?

ved-rivos, 2025-05-07:
This requirement states that external interrupts generated by iommu itself must use MSI and may use WSI

Chen-rui55, 2025-05-07:
> This requirement states that “external interrupts originated by the iommu itself” must use MSI and may use WSI

Does the external interrupt here refer to command queue interrupt, fault queue interrupt, performance monitoring interrupt, and other interrupts ?

ved-rivos, 2025-05-07:
Correct . These are interrupts from the command queue , fault queue , page request queue , and the hardware performance monitor .

Chen-rui55, 2025-05-08:
> Correct . These are interrupts from the command queue , fault queue , page request queue , and the hardware performance monitor .

Thanks. You mentioned that it is necessary to support MSI and be able to support WSI. Why must MSI be used and WSI may be used. What is the reason for this. 
In addition, I noticed that the code of iommu driver currently only supports WSI, only be set to MSI when iommu is considered a PCIe peripheral.

/* =============================================  from the iommu driver code
*/
	/* For now we only support WSI */
	switch (FIELD_GET(RISCV_IOMMU_CAPABILITIES_IGS, iommu->caps)) {
	case RISCV_IOMMU_CAPABILITIES_IGS_WSI:
	case RISCV_IOMMU_CAPABILITIES_IGS_BOTH:
		break;
	default:
		return dev_err_probe(dev, -ENODEV,
				     "unable to use wire-signaled interrupts\n");
	}
	.......
	/* Enable wire-signaled interrupts, fctl.WSI */
	if (!(iommu->fctl & RISCV_IOMMU_FCTL_WSI)) {
		iommu->fctl |= RISCV_IOMMU_FCTL_WSI;
		riscv_iommu_writel(iommu, RISCV_IOMMU_REG_FCTL, iommu->fctl);
	}

jones-drew, 2025-05-08:
> In addition, I noticed that the code of iommu driver currently only supports WSI, only be set to MSI when iommu is considered a PCIe peripheral.
> 

You must be looking at old code. commit d5f88acdd6ff ("iommu/riscv: Add support for platform msi") was merged for v6.14-rc1

Chen-rui55, 2025-05-08:
Thanks, it is indeed old code. I still don't understand why  servers-soc must use MSI or may use WSI.  Can we just use WSI? Is MSI more flexible and easy to expand, or is it better to use with AIA(Advanced Interrupt Architecture) on  servers-soc.

closed 2025-05-14 by ved-rivos